MAC-VO/MAC-VO #4: Visualization questions

zhanghanduo, 2024-11-21:
Thanks for the great work. Just want to ask about the visualization of the trajectory and 3D construction of the point cloud. When running `$ python -m Scripts.AdHoc.DemoSequence --space [RESULT_PATH] --data [DATA_CONFIG_PATH]`, I can only see temporary local  sparse point cloud. Maybe not familiar with rerun, how do I visualize the consistent global map?

MarkChenYutian, 2024-11-23:
Hi,

To visualize the point cloud of all frames simultaneously, you can set the "Visible Time Range" of point cloud instance in the following steps

1. Open the right panel of the Rerun visualizer
2. Select the point cloud instance on the left (names may be different from the screenshot as we are actively updating our visualization used internally)
3. Find the `Visible Time Range` option and expand the menu
4. Set Visible time range to `Override` and select `beginning of timeline`

Feel free to follow up if you have any additional question : )

[image]
s multiple frames,

zhanghanduo, 2024-11-24:
Thank you for your detailed explanation and guidance! I truly appreciate that. This has clarified everything.

closed 2024-11-24 by zhanghanduo