MAC-VO/MAC-VO #5: Question about the change from tensor_map.pth to tensor_map.npz and visualization code

lingshijiang, 2025-03-24:
I have a question regarding the recent changes to the serialization format of the tensor_map. I noticed that the project previously generated tensor_map.pth files, but now it seems to be generating tensor_map.npz files.

Could you please explain the reason behind this change? What are the advantages of using .npz over .pth in this case?
I've noticed an inconsistency in how the tensor_map is saved and loaded.

In interface.py, the tensor_map is saved as an .npz file using np.savez_compressed:
np.savez_compressed(saveto.path("tensor_map.npz"), **global_map.serialize())
However, in DemoSequence.py, the code attempts to load a .pth file using np.load:
# DemoSequence.py
macvo_map = VisualMap.deserialize(np.load(space.path("tensor_map.pth")))

This inconsistency leads to an error when trying to load a previously saved tensor_map. It seems like either the saving or loading part needs to be updated to ensure consistency. Could you please clarify which format (.pth or .npz) is intended to be used, and update the code accordingly?

Thank you for your time and assistance.

MarkChenYutian, 2025-03-26:
Hi, @lingshijiang 

Thanks for raising this — happy to clarify the motivation behind the switch from .pth to .npz.

Previously, we saved the entire VisualMap object using torch.save to a .pth file, which relies on pickle for serialization. This tightly couples the saved file with the Python class definition, making it hard for other projects to load the map unless they also have the VisualMap class defined — otherwise, pickle throws an error.

To improve interoperability (especially for downstream projects that use the tensor_map), we switched to the .npz format, which stores the map data as a plain dictionary of NumPy arrays. This makes the file much easier to load and use without needing the original class or source code.

---

As for the save/load inconsistency you mentioned - this is a bug we missed when upgrading our save/load protocol. We used automated testing & static analysis to try detecting such bugs but this one is missed out from our CI/CD. I will fix this later this week. Thanks for reporting!

MarkChenYutian, 2025-03-26:
To fix this problem, please use the `deserialize` method in the `VisualMap class` to load the VisualMap from dumped file. 

https://github.com/MAC-VO/MAC-VO/blob/bebcb01c1428dd99038c8e480113a7dee0d4d890/Module/Map/VisualMap.py#L118C9-L118C20

You are also welcomed to open up a PR for this if you have time! 🙌

haleqiu, 2025-04-15:
@lingshijiang Thank you for pointing out this. I close this issue as there are no follow-up discussion.

closed 2025-04-15 by haleqiu