MAC-VO/MAC-VO #6: What is the intended License for this repository?

jatkinson-CRL, 2025-03-25:
Hello, your work here is extremely impressive! Thank you for sharing it with the world. As I was looking through your repository, I noticed that the BSD 3-Clause LICENSE file was removed in the same commit that added the dense mapping feature (https://github.com/MAC-VO/MAC-VO/commit/6899371629b5e11041b5a5315f03fc148cab4e47#diff-c693279643b8cd5d248172d9c22cb7cf4ed163a3c98c8a3f69c2717edd3eacb7L1). 

However, your README still says that the repository is licensed under the BSD 3-Clause license (https://raw.githubusercontent.com/MAC-VO/MAC-VO/refs/heads/main/README.md). Since it is still in the README as BSD-3, I was not sure if the LICENSE file was removed by mistake in that aforementioned commit.

I was hoping that you could clarify what the intended license of this repository is after the addition of the dense mapping feature. If it is still intended to be licensed under the BSD 3-Clause license or any other license, would you mind adding the intended LICENSE file back to the repository so that others know the terms/conditions under which they can build upon the new dense mapping features?

Looking forward to hearing back from you, and thank you again for sharing this excellent work!

haleqiu, 2025-03-26:
@jatkinson-CRL Thank you for pointing this out. It is the BSD 3-Clause license

closed 2025-03-26 by haleqiu

MarkChenYutian, 2025-05-29:
As an update, we have changed the license to MIT license since it is a more preferable license for Carnegie Mellon University.