MAC-VO/MAC-VO #8: 'flow' folder does not provide in abf001_select demo dataset

Xie-HIT, 2025-04-25:
I download the demo dataset from the provided [link](https://drive.google.com/file/d/1kCTNMW2EnV42eH8g2STJHcVWEbVKbh_r/view?usp=sharing). However, the dataset does not contain a folder called 'flow'.

When running:
`python MACVO.py --odom Config/Experiment/MACVO/MACVO.yaml --data Config/Sequence/TartanAir_abandonfac_001.yaml`

I got the following:
`AssertionError: Optical flow directory /home/xiechen/MAC-VO/Dataset/abf001_select/P001_select/flow does not exist`

MarkChenYutian, 2025-04-28:
Hi, the flow directory is not required to run the demo, please follow the updated instruction in PR #7 *or* change the `gtFlow` in `TartanAir_abandonfac_001.yaml` to `false` when running the script.

MarkChenYutian, 2025-04-28:
I will mark this issue as resolved, please feel free to reopen if the above method didn't resolve the error!

closed 2025-04-28 by MarkChenYutian

Xie-HIT, 2025-04-29:
Thanks for your quick reply! Following your instructions, I successfully run the demo of MAC-VO after downloading the `dev/fixgit` branch. However, I stuck at visualizing. Since I want to look the eye-catcher figure, I change the code in Scripts/AdHoc/DemoSequence.py following #5:
`macvo_map = VisualMap.deserialize(np.load(space.path("tensor_map.npz")))`. But I get a new error about T_Fields. To solve it, I further change the deserialize function of TensorBundle (in Module/Map/Graph.py) from:
`
@classmethod
def deserialize(cls, prefix: str, value: dict[str, np.ndarray]) -> Self:
        allowed_fields: T_Fields | None = T.get_args(cls.__orig_bases__[0].__args__[0])  # type: ignore
        if not allowed_fields:
            raise Exception("T_Fields must be specified with concrete string literal types.")
        
        data: dict[T_Fields, torch.Tensor] = T.cast(dict[T_Fields, torch.Tensor], {
            k : torch.tensor(value[f"{prefix}/k"])
            for k in allowed_fields
        })
        return cls.init(data)  
`
to
`
def deserialize(self, prefix: str, value: dict[str, np.ndarray]) -> Self:
        allowed_fields: T_Fields | None = T.get_args(self.__dict__['__orig_class__'].__args__[0])  # type: ignore
        if not allowed_fields:
            raise Exception("T_Fields must be specified with concrete string literal types.")
        
        data: dict[T_Fields, torch.Tensor] = T.cast(dict[T_Fields, torch.Tensor], {
            k : torch.tensor(value[f"{prefix}/{k}"])
            for k in allowed_fields
        })
        return self.init(data)
`. 
Then I  run the following command:
`python -m Scripts.AdHoc.DemoSequence --space ../../Results/MACVO@abf001/04_29_063404 --data ../../Config/Sequence/TartanAir_example.yam --from_frame 0 --to_frame <NUM>` (for example, NUM=100)
But I still get an error:
`
Traceback (most recent call last):
  File "/home/xiechen/MAC-VO/Scripts/AdHoc/DemoSequence.py", line 74, in <module>
    visualize_run(SequenceBase.instantiate(datacfg.type, datacfg.args), Sandbox.load(args.space), args.from_frame, args.to_frame)
  File "/home/xiechen/MAC-VO/Scripts/AdHoc/DemoSequence.py", line 31, in visualize_run
    curr_obs    = macvo_map.get_frame2match(curr_frame)
  File "/home/xiechen/MAC-VO/Module/Map/VisualMap.py", line 87, in get_frame2match
    return self.match[self.frame2match.project(frame.index)]
  File "/home/xiechen/MAC-VO/Module/Map/Graph.py", line 183, in project
    offsets = self.ranges[from_index, :, 0].flatten()
  File "/home/xiechen/MAC-VO/Utility/Extensions/TensorExtension.py", line 135, in __getitem__
    def __getitem__(self, slice): return self.tensor.__getitem__(slice)
IndexError: index is out of bounds for dimension with size 0

Process finished with exit code 1
`
Now I have no idea to visualize the 3D trajectory. I would appreciate further guidance on this matter.