MAC-VO/MAC-VO #11: Operational issues

xl232lx, 2025-05-27:
Hello author, may I ask how I test the recorded bag file containing the topic of cameras？

haleqiu, 2025-05-28:
@xl232lx Hi, what datasets are you refering to?

xl232lx, 2025-05-29:
@haleqiu Hello, currently I'm running a demo with the abf001_select data. I want to build a dense map. Could you please tell me how to combine rerun to generate a dense point cloud and save the map? By the way, are there any links to the Zed, VBR, and EuRoC datasets?

haleqiu, 2025-05-29:
@xl232lx 
Please check the README.md for the mapping mode setup, you need to change the config.

VBR and EuRoC are open-access datasets, Please refer to their repo/website for the download.

ZED dataset is something we collected in our office. We can realease them if neccessary, but we have no ground truth. Do you need the Zed data for evaluation?

xl232lx, 2025-05-29:
@haleqiu Hello. Thank you for your patient answers. I'm not very familiar with rerun, so running the demo didn't go very smoothly. I run 'python3 MACVO. py - odom Config/Experiment/MACVO/MACVO_example yaml - data Config/Sequence/TartanAir_example yaml --useRR'
After this command, I didn't bring up the dense map window, so I asked this question. I'll read the readme carefully first and run it with the open-source dataset.

MarkChenYutian, 2025-05-29:
Hi @xl232lx , for the dense point cloud visualization, please refer to this issue
* #4

xl232lx, 2025-05-30:
@MarkChenYutian ok，thanks!

closed 2025-05-30 by MarkChenYutian