MAC-VO/MAC-VO #18: About the Runtime Analysis

panxkun, 2025-06-08:
Hello! I am very interested in this solid work! And I configured and ran it very smoothly.

I have a question about the Runtime Analysis in paper. The raw fps is about 2fps, is it the default configuration of this repo now?

And I also wonder what is the bottleneck of the system performance now. I found that the TensorRT version's efficiency is only slightly better than the raw version, which is quite different from the significant speedup I learned about from the official TensorRT sources compared to the PyTorch version. 

The appendix mentions the fast mode and states that it achieves 70% of the performance. Does this 70% performance refer only to the pose accuracy measured by RMSE, meaning that the accuracy decreases as the performance (speed) is improved?

MarkChenYutian, 2025-06-08:
Hi, thanks for the great question! System speed is something we care a lot about, and here are some quick clarifications:

 1.	The default MAC-VO config uses multiprocessing (MP) to parallelize backend optimization (CPU) with the frontend (GPU), as noted in Table 5 and Appendix I. This gives a significant speed boost over the single-process baseline (Raw).

 2.	The original FlowFormer code ([official repo](https://github.com/drinkingcoder/FlowFormer-Official)) used in the paper has poor engineering practices—e.g., host-device syncs and dynamically allocated constants—that hinder TensorRT optimization.We’ve since re-implemented FlowFormer ([our repo](https://github.com/MAC-VO/S_FlowFormer)) to clean up redundant operations. Our version runs ~2× faster and shows a more noticeable acceleration under TensorRT.

    That said, even with our optimized version, TensorRT can only safely compile at optimization level 1 (out of 5). Higher levels cause significant accuracy degradation—likely due to aggressive fusions that harm numeric precision. This may be because TensorRT is better tuned for LM tasks (e.g., logits in [0,1]) rather than optical flow (values up to hundreds).

 3.	The “fast mode” reduces iterative refinement steps from 12 → 4, which improves runtime but sacrifices some pose accuracy (as reflected in higher RTE and ROE). So yes, the “70% performance” refers to reduced accuracy, not runtime.

panxkun, 2025-06-08:
Great, thanks for your reply!

closed 2025-06-08 by panxkun

panxkun, 2025-06-30:
Thanks, it helps a lot!