MAC-VO/MAC-VO #19: About MAC-VO Fast/TensorRT/Defualt config

chi-qhsun, 2025-06-09:
Hi, thanks for bringing out the solid work. I successfully ran MACVO with self-collected Zed Data with:

--odom MACVO_example.yaml
--data Zed.yaml(customized)

I wonder how I can know which model I'm using, and how I can switch between [default/fast/tensorrt] architectures.

I'd appreciate your help, thanks for your great work!

haleqiu, 2025-06-10:
@chi-qhsun For now the TensorRT mode is officially supported. You may refer to #18 .

MarkChenYutian, 2025-06-11:
As a follow-up, the MAC-VO you are running is completely defined by the `yaml` file passed in as the odometry config. The default `example` config uses `CUDAGraph_FlowFormerCov` frontend for instance, you can switch to a slower version by replacing the `type` to `FlowFormerCov`.

chi-qhsun, 2025-06-11:
Thanks for the clarification. I will focus on the default setting since the TensorRT version is not officially supported yet.

closed 2025-06-11 by chi-qhsun

chi-qhsun, 2025-06-30:
Thanks for your great work. Will check out soon!