MAC-VO/MAC-VO #21: the code generates the two-dimensional uncertainty for pixel points

aidadgy, 2025-06-24:
Hello author, could you please tell me which part of the code generates the two-dimensional uncertainty for pixel points, or specifically which section handles the construction of the three-dimensional covariance matrix? I'd appreciate it if you could point me directly to the relevant code. Looking forward to your reply

MarkChenYutian, 2025-06-30:
Hi @aidadgy , the code for projecting 2D uncertainty to 3D is in this file: `Module/Covariance/Project2to3.py`. There are multiple implementations in this file and specifically, MAC-VO is using the `MatchCovariance` class for implementation

https://github.com/MAC-VO/MAC-VO/blob/65807db68d8c20f77c5bef114cc8fb989b99f268/Module/Covariance/Project2to3.py#L113-L190

The config for controlling which implementation to use is here:

https://github.com/MAC-VO/MAC-VO/blob/65807db68d8c20f77c5bef114cc8fb989b99f268/Config/Experiment/MACVO/MACVO.yaml#L30-L38

MarkChenYutian, 2025-07-04:
Hi @aidadgy , I will close this issue due to inactive, feel free to followup if you have any questions!

aidadgy, 2025-08-02:
> Hi [@aidadgy](https://github.com/aidadgy) , the code for projecting 2D uncertainty to 3D is in this file: `Module/Covariance/Project2to3.py`. There are multiple implementations in this file and specifically, MAC-VO is using the `MatchCovariance` class for implementation嗨，@Aidadgy，将2D不确定性投射到3D的代码在此文件中：模块/协方差/project2to3.py。该文件中有多个实现，具体来说，Mac-Vo正在使用MatchCovariancer类进行实施
> 
> [MAC-VO/Module/Covariance/Project2to3.pymac-vo/模块/协方差/project2to3.py](https://github.com/MAC-VO/MAC-VO/blob/65807db68d8c20f77c5bef114cc8fb989b99f268/Module/Covariance/Project2to3.py#L113-L190)
> 
> Lines 113 to 190 in [65807db](/MAC-VO/MAC-VO/commit/65807db68d8c20f77c5bef114cc8fb989b99f268)第113至190行65807DB
>  class MatchCovariance(ICovariance2to3): 
> 类MatchCoariance（ICOVARIANCE2TO3）：
>      """”“” 
>      Covariance model used by MAC-VO. See section III.C for detail.Mac-Vo使用的协方差模型。有关详细信息，请参见第三节。 
>      """”“” 
>      def __init__(self, config: SimpleNamespace): 
> def __init __（self，config：simpleenamespace）：
>          super().__init__(config) 
> super（）.__ init __（config）
>          self.config.kernel_size_hlf = self.config.kernel_size // 2
> self.config.kernel_size_hlf = self.config.kernel_size // 2
>  
>   
>      @Timer.cpu_timeit("Cov Model")@Timer.cpu_timeit（“ COV模型”） 
>      @Timer.gpu_timeit("Cov Model")@Timer.gpu_timeit（“ COV模型”） 
>      def estimate(self, frame: StereoData, kp: torch.Tensor, depth_est: IStereoDepth.Output, depth_cov: torch.Tensor | None, flow_cov: torch.Tensor | None) -> torch.Tensor:         
> def估计（自我，框架：stereodata，kp：torch.tensor，depth_est：istereodepth.output，depth_cov：torch.tensor | none，flow_cov：flow_cov：torch.tensor | none） - > torch.tensor：> torch.tensor：
>          n_sample = kp.size(0) 
> n_sample = kp.size（0）
>   
>          kp_long = kp.clone().long() 
> kp_long = kp.clone（）。long（）
>          if (has_flow_cov_flag := flow_cov is not None): 
> 如果（has_flow_cov_flag：= flow_cov不是无）：
>              # Min clamp to 0.16 since camera plane is descretelized by pixels, which has at least an＃min夹至0.16 
>              # uncertainty of 0.16＃0.16的不确定性 
>              flow_cov[..., :2].clamp_(min=self.config.min_flow_cov**2) 
> flow_cov [...，：2] .clamp_（min = self.config.min_flow_cov ** 2）
>          else: 
> 别的：
>              flow_cov = torch.ones((n_sample, 3), device=self.config.device, dtype=torch.float) * self.config.match_cov_default
> flow_cov = torch.ones（（（n_sample，3），device = self.config.device，dtype = type = terch.float） * self.config.match_cov_default
>  
>              assert flow_cov is not None
> 断言Flow_cov不是没有
>  
>              flow_cov[..., 2] = 0.
> flow_cov [...，2] = 0。
>  
>   
>          var_u, var_v, var_uv = flow_cov[..., 0], flow_cov[..., 1], flow_cov[..., 2] 
> var_u，var_v，var_uv = flow_cov [...，0]，flow_cov [...，1]，flow_cov [...，2]
>          kp_u, kp_v = kp[..., 0], kp[..., 1] 
> kp_u，kp_v = kp [...，0]，kp [...，1]
>   
>          # Get local depth average and variance＃获取本地深度和差异 
>          u_indices = torch.arange( 
> u_indices = torch.arange（
>              -self.config.kernel_size_hlf, self.config.kernel_size_hlf + 1, dtype=torch.long, device=self.config.device
> -self.config.kernel_size_hlf，self.config.kernel_size_hlf + 1，dtype = type = torch.long，device = self.config.device.device.device.device
>  
>          ) 
>          v_indices = torch.arange( 
>              -self.config.kernel_size_hlf, self.config.kernel_size_hlf + 1, dtype=torch.long, device=self.config.device 
>          ) 
>          uu, vv = torch.meshgrid(u_indices, v_indices, indexing="ij") 
>   
>          all_u_indices = kp_long[:, 0].unsqueeze(-1) + uu.reshape(1, -1) 
>          all_v_indices = kp_long[:, 1].unsqueeze(-1) + vv.reshape(1, -1) 
>   
>          cov_matrices = create_2x2_matrix([[var_u, var_uv], [var_uv, var_v]], n_sample=n_sample, device=self.config.device) 
>          local_filters = gaussain_full_kernels(cov_matrices, kernel_size=self.config.kernel_size) 
>           
>          patches = depth_est.depth[..., all_v_indices, all_u_indices].view( 
>              n_sample, self.config.kernel_size, self.config.kernel_size 
[... 40 lines cut ...]
> Lines 30 to 38 in [65807db](/MAC-VO/MAC-VO/commit/65807db68d8c20f77c5bef114cc8fb989b99f268)第30至38行65807DB
>  cov: 
> COV：
>    obs: 
> obs：
>      type: MatchCovariance
> 类型：MatchCovariance
>  
>      args: 
> args：
>        device: *device
> 设备： *设备
>  
>        kernel_size: 31
> kernel_size：31
>  
>        match_cov_default: 0.25
> match_cov_default：0.25
>  
>        min_depth_cov: 0.05
> min_depth_cov：0.05
>  
>        min_flow_cov: 0.25
> min_flow_cov：0.25
I am glad to receive the author's response. Here, kp_u, kp-v, and var_u, var-v, and var_uv in the code represent keypoints and their covariance, which can be understood this way. However, is the selection of keypoints done before covariance estimation? I think the meaning of the paper is to estimate the covariance of all pixels first, and then select keypoints. Who comes first and who comes last? Thank you for your answer

aidadgy, 2025-08-02:
> Hi [@aidadgy](https://github.com/aidadgy) , the code for projecting 2D uncertainty to 3D is in this file: `Module/Covariance/Project2to3.py`. There are multiple implementations in this file and specifically, MAC-VO is using the `MatchCovariance` class for implementation
> 
> [MAC-VO/Module/Covariance/Project2to3.py](https://github.com/MAC-VO/MAC-VO/blob/65807db68d8c20f77c5bef114cc8fb989b99f268/Module/Covariance/Project2to3.py#L113-L190)
> 
> Lines 113 to 190 in [65807db](/MAC-VO/MAC-VO/commit/65807db68d8c20f77c5bef114cc8fb989b99f268)
>  class MatchCovariance(ICovariance2to3): 
>      """ 
>      Covariance model used by MAC-VO. See section III.C for detail. 
>      """ 
>      def __init__(self, config: SimpleNamespace): 
>          super().__init__(config) 
>          self.config.kernel_size_hlf = self.config.kernel_size // 2 
>   
>      @Timer.cpu_timeit("Cov Model") 
>      @Timer.gpu_timeit("Cov Model") 
>      def estimate(self, frame: StereoData, kp: torch.Tensor, depth_est: IStereoDepth.Output, depth_cov: torch.Tensor | None, flow_cov: torch.Tensor | None) -> torch.Tensor:         
>          n_sample = kp.size(0) 
>   
>          kp_long = kp.clone().long() 
>          if (has_flow_cov_flag := flow_cov is not None): 
>              # Min clamp to 0.16 since camera plane is descretelized by pixels, which has at least an 
>              # uncertainty of 0.16 
>              flow_cov[..., :2].clamp_(min=self.config.min_flow_cov**2) 
>          else: 
>              flow_cov = torch.ones((n_sample, 3), device=self.config.device, dtype=torch.float) * self.config.match_cov_default 
>              assert flow_cov is not None 
>              flow_cov[..., 2] = 0. 
>   
>          var_u, var_v, var_uv = flow_cov[..., 0], flow_cov[..., 1], flow_cov[..., 2] 
>          kp_u, kp_v = kp[..., 0], kp[..., 1] 
>   
>          # Get local depth average and variance 
>          u_indices = torch.arange( 
>              -self.config.kernel_size_hlf, self.config.kernel_size_hlf + 1, dtype=torch.long, device=self.config.device 
>          ) 
>          v_indices = torch.arange( 
>              -self.config.kernel_size_hlf, self.config.kernel_size_hlf + 1, dtype=torch.long, device=self.config.device 
>          ) 
>          uu, vv = torch.meshgrid(u_indices, v_indices, indexing="ij") 
>   
>          all_u_indices = kp_long[:, 0].unsqueeze(-1) + uu.reshape(1, -1) 
>          all_v_indices = kp_long[:, 1].unsqueeze(-1) + vv.reshape(1, -1) 
>   
>          cov_matrices = create_2x2_matrix([[var_u, var_uv], [var_uv, var_v]], n_sample=n_sample, device=self.config.device) 
>          local_filters = gaussain_full_kernels(cov_matrices, kernel_size=self.config.kernel_size) 
>           
>          patches = depth_est.depth[..., all_v_indices, all_u_indices].view( 
>              n_sample, self.config.kernel_size, self.config.kernel_size 
>          ).to(self.config.device) 
>          patches = patches.permute(0, 2, 1) 
>   
>          # Weighted Average 
>          wavg_depth = (local_filters * patches).sum(dim=[1, 2]) 
>          if (has_flow_cov_flag or (depth_cov is None)): 
>              # Weighted Variance 
>              wvar_depth = torch.sum( 
>                  local_filters * (patches - (wavg_depth.unsqueeze(1).unsqueeze(1))).square(), 
>                  dim=[1, 2], 
>              ) 
>          else: 
>              assert depth_cov is not None 
>              wvar_depth = depth_cov 
>           
>          wvar_depth = wvar_depth.clamp(min=self.config.min_depth_cov) 
>   
>          # Inverse project 2D keypoint to 3D space. 
>          cov = Covariance_2to3_full( 
>              var_u, var_uv, var_v, wvar_depth, 
>              kp_u, kp_v, wavg_depth, 
>              frame.fx, frame.fy, frame.cx, frame.cy 
>          ).double() 
>   
>          return cov 
>       
>      @classmethod 
>      def is_valid_config(cls, config: SimpleNamespace | None) -> None: 
>          cls._enforce_config_spec(config, { 
>              "kernel_size"       : lambda k: isinstance(k, int) and k > 0 and (k % 2 == 1), 
>              "match_cov_default" : lambda c: isinstance(c, (int, float)) and c > 0., 
>              "min_flow_cov"      : lambda c: isinstance(c, (int, float)) and c > 0., 
>              "min_depth_cov"     : lambda c: isinstance(c, (int, float)) and c > 0., 
>              "device"            : lambda dev: isinstance(dev, str) and (("cuda" in dev) or (dev == "cpu")) 
>          }) 
> 
> The config for controlling which implementation to use is here:
> 
> [MAC-VO/Config/Experiment/MACVO/MACVO.yaml](https://github.com/MAC-VO/MAC-VO/blob/65807db68d8c20f77c5bef114cc8fb989b99f268/Config/Experiment/MACVO/MACVO.yaml#L30-L38)
> 
> Lines 30 to 38 in [65807db](/MAC-VO/MAC-VO/commit/65807db68d8c20f77c5bef114cc8fb989b99f268)
>  cov: 
>    obs: 
>      type: MatchCovariance 
>      args: 
>        device: *device 
>        kernel_size: 31 
>        match_cov_default: 0.25 
>        min_depth_cov: 0.05 
>        min_flow_cov: 0.25

I am glad to receive the author's response. Here, kp_u, kp-v, and var_u, var-v, and var_uv in the code represent keypoints and their covariance, which can be understood this way. However, is the selection of keypoints done before covariance estimation? I think the meaning of the paper is to estimate the covariance of all pixels first, and then select keypoints. Who comes first and who comes last? Thank you for your answer