MAC-VO/MAC-VO #24: Runtime Error CUDA/Torch

ddkats, 2025-07-07:
Hi @haleqiu  

Congrats on your excellent work! I'm very interested in testing it as part of a larger mapping framework using ZED cameras.

Following your instructions step by step using the Docker setup, I encountered a runtime **CUDA/Torch issue**. Do you know what might be causing this? I’ve tested it on both Ubuntu 20 and 22 with CUDA versions 11.4, 11.7, and 12.2, respectively. Should I switch the host to CUDA 12.4 to run it successfully? Have you ever tested it with earlier CUDA versions?

```
RuntimeError: Attempting to deserialize object on a CUDA device but torch.cuda.is_available() is False. If you are running on a CPU-only machine, please use torch.load with map_location=torch.device('cpu') to map your storages to the CPU.
```

The entire log of the error is this:
```
macvo@8abe8508773e:~/workspace$ sudo python3 MACVO.py --odom Config/Experiment/MACVO/MACVO_Performant.yaml --data Config/Sequence/TartanAir_example.yaml
[08:45:46] INFO     PID 67 generated new fontManager                            font_manager.py:1639
Initialized new /root/.evo/settings.json
fatal: detected dubious ownership in repository at '/home/macvo/workspace'
To add an exception for this directory, call:

	git config --global --add safe.directory /home/macvo/workspace
/usr/local/lib/python3.10/dist-packages/torch/cuda/__init__.py:174: UserWarning: CUDA initialization: Unexpected error from cudaGetDeviceCount(). Did you run some cuda functions before calling NumCudaDevices() that might have already set an error? Error 804: forward compatibility was attempted on non supported HW (Triggered internally at /pytorch/c10/cuda/CUDAFunctions.cpp:109.)
  return torch._C._cuda_getDeviceCount() > 0
Traceback (most recent call last):
  File "/home/macvo/workspace/MACVO.py", line 140, in <module>
    system = MACVO[StereoFrame].from_config(asNamespace(exp_space.config))
  File "/home/macvo/workspace/Odometry/MACVO.py", line 84, in from_config
    Frontend            = Module.IFrontend.instantiate(odomcfg.frontend.type, odomcfg.frontend.args)
  File "/home/macvo/workspace/Utility/Extensions/SubclassRegistry.py", line 26, in instantiate
    return cls.get_class(type)(*args, **kwargs)
  File "/home/macvo/workspace/Module/Frontend/Frontend.py", line 270, in __init__
    super().__init__(config)
  File "/home/macvo/workspace/Module/Frontend/Frontend.py", line 172, in __init__
    ckpt  = torch.load(self.config.weight, map_location=self.config.device, weights_only=True)
  File "/usr/local/lib/python3.10/dist-packages/torch/serialization.py", line 1516, in load
    return _load(
  File "/usr/local/lib/python3.10/dist-packages/torch/serialization.py", line 2114, in _load
    result = unpickler.load()
  File "/usr/local/lib/python3.10/dist-packages/torch/_weights_only_unpickler.py", line 532, in load
    self.append(self.persistent_load(pid))
  File "/usr/local/lib/python3.10/dist-packages/torch/serialization.py", line 2078, in persistent_load
    typed_storage = load_tensor(
  File "/usr/local/lib/python3.10/dist-packages/torch/serialization.py", line 2044, in load_tensor
    wrap_storage = restore_location(storage, location)
  File "/usr/local/lib/python3.10/dist-packages/torch/serialization.py", line 1854, in restore_location
    return default_restore_location(storage, map_location)
  File "/usr/local/lib/python3.10/dist-packages/torch/serialization.py", line 698, in default_restore_location
    result = fn(storage, location)
  File "/usr/local/lib/python3.10/dist-packages/torch/serialization.py", line 636, in _deserialize
    device = _validate_device(location, backend_name)
  File "/usr/local/lib/python3.10/dist-packages/torch/serialization.py", line 605, in _validate_device
    raise RuntimeError(
RuntimeError: Attempting to deserialize object on a CUDA device but torch.cuda.is_available() is False. If you are running on a CPU-only machine, please use torch.load with map_location=torch.device('cpu') to map your storages to the CPU.
```

Thanks!

MarkChenYutian, 2025-07-07:
Hi, our implementation should not be strongly bound to a CUDA runtime since we haven't written any CUDA kernels (yet). From the exception, it seems like `torch.cuda.is_available()` is `False` in your Docker environment. Did you install the Nvidia Container Runtime on your machine?

Our code *probably* works in a CPU-only environment (runs very slowly, though). You can try CPU-only mode by switching `device` to `cpu` in `MACVO_Performant.yaml`.

MarkChenYutian, 2025-07-08:
Hi @ddkats 

I still suspect there is a configuration problem with the docker container, since the last line of previous exception:

```
RuntimeError: Attempting to deserialize object on a CUDA device but torch.cuda.is_available() is False.
If you are running on a CPU-only machine, please use torch.load with map_location=torch.device('cpu') to map your storages to the CPU.
```

We are developing using a minimal PyTorch version of 2.4, which depends on CUDA runtime 12.4+. Our docker file also uses CUDA 12.4 for that reason.

https://github.com/MAC-VO/MAC-VO/blob/b72acaa3c4c64c05f04847263c8ea1defc4fd5ed/Docker/Dockerfile#L1

Since the CUDA calls in Docker will be "forwarded" to the host machine CUDA (image from https://github.com/NVIDIA/nvidia-container-toolkit), I would recommend upgrading the CUDA runtime on Host machine, if possible and see if the `nvidia-smi` is working in Docker container. (`nvcc` in container may still show version while `CUDA` is actually not usable).

[image]

As for the CPU-only runtime, yes the CUDA graph acceleration only works on CUDA GPU. Please change `type: CUDAGraph_FlowFormerCovFrontend` in the config file (Under `Odometry.Frontend`) to `type: FlowFormerCovFrontend` to use vanilla network.

ddkats, 2025-07-09:
Hi @MarkChenYutian 

Thanks for your time and your efforts to debug this issue, I really appreciate it.

The nvidia-smi from inside the docker env works just fine. Also I updated the nvidia-toolkit in the host pc and it now runs toolkit 12.4. So the host has nvidia-smi=12.2 (can't update the drivers) and nvcc=12.4, the error is the same after this update. 

Regarding the CPU-mode even after setting the ```Odometry.Frontend``` to type: ```FlowFormerCovFrontend``` + ```device cpu```, there is still CUDA-related errors. 

The ```nvidia-smi``` from inside the Docker environment works just fine. I also updated the NVIDIA toolkit on the host PC, which now runs Toolkit 12.4. So, the host has ```nvidia-smi = 12.2``` and ```nvcc = 12.4```. The error remains the same after this update (error (1)).

Regarding CPU mode, even after setting the ```Odometry.Frontend``` to type: ```FlowFormerCovFrontend``` + ```device: cpu```, there are still CUDA-related errors (error (2)).

The full logs for both of error (1) and error (2) are outlined below, respectively.

error (1)
```
macvo@a95aff6f37a9:~$ nvidia-smi
Wed Jul  9 07:16:36 2025       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.183.01             Driver Version: 535.183.01   CUDA Version: 12.4     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 2080 ...    Off | 00000000:08:00.0  On |                  N/A |
|  0%   53C    P8              12W / 250W |    571MiB /  8192MiB |      2%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
+---------------------------------------------------------------------------------------+

macvo@63fe2dc8f94d:~/workspace$ sudo python3 MACVO.py --odom Config/Experiment/MACVO/MACVO_Performant.yaml --data Config/Sequence/TartanAir_example.yaml
[09:29:40] INFO     PID 65 generated new fontManager                                                                     font_manager.py:1639
Initialized new /root/.evo/settings.json
fatal: detected dubious ownership in repository at '/home/macvo/workspace'
To add an exception for this directory, call:

	git config --global --add safe.directory /home/macvo/workspace
/usr/local/lib/python3.10/dist-packages/torch/cuda/__init__.py:174: UserWarning: CUDA initialization: Unexpected error from cudaGetDeviceCount(). Did you run some cuda functions before calling NumCudaDevices() that might have already set an error? Error 804: forward compatibility was attempted on non supported HW (Triggered internally at /pytorch/c10/cuda/CUDAFunctions.cpp:109.)
  return torch._C._cuda_getDeviceCount() > 0
Traceback (most recent call last):
  File "/home/macvo/workspace/MACVO.py", line 140, in <module>
    system = MACVO[StereoFrame].from_config(asNamespace(exp_space.config))
  File "/home/macvo/workspace/Odometry/MACVO.py", line 84, in from_config
    Frontend            = Module.IFrontend.instantiate(odomcfg.frontend.type, odomcfg.frontend.args)
  File "/home/macvo/workspace/Utility/Extensions/SubclassRegistry.py", line 26, in instantiate
    return cls.get_class(type)(*args, **kwargs)
  File "/home/macvo/workspace/Module/Frontend/Frontend.py", line 270, in __init__
    super().__init__(config)
  File "/home/macvo/workspace/Module/Frontend/Frontend.py", line 172, in __init__
    ckpt  = torch.load(self.config.weight, map_location=self.config.device, weights_only=True)
  File "/usr/local/lib/python3.10/dist-packages/torch/serialization.py", line 1516, in load
    return _load(
  File "/usr/local/lib/python3.10/dist-packages/torch/serialization.py", line 2114, in _load
    result = unpickler.load()
  File "/usr/local/lib/python3.10/dist-packages/torch/_weights_only_unpickler.py", line 532, in load
    self.append(self.persistent_load(pid))
  File "/usr/local/lib/python3.10/dist-packages/torch/serialization.py", line 2078, in persistent_load
    typed_storage = load_tensor(
  File "/usr/local/lib/python3.10/dist-packages/torch/serialization.py", line 2044, in load_tensor
    wrap_storage = restore_location(storage, location)
  File "/usr/local/lib/python3.10/dist-packages/torch/serialization.py", line 1854, in restore_location
    return default_restore_location(storage, map_location)
  File "/usr/local/lib/python3.10/dist-packages/torch/serialization.py", line 698, in default_restore_location
[... 54 lines cut ...]
    self.run_pair(self.prev_keyframe[0], frame)
[... cut ...]
  File "/home/macvo/workspace/Odometry/Interface.py", line 67, in receive_frames
    self.terminate()
  File "/home/macvo/workspace/Odometry/MACVO.py", line 376, in terminate
    self.Optimizer.write_map(self.graph)
  File "/home/macvo/workspace/Module/Optimization/Interface.py", line 227, in write_map
    graph_res_local = self.__get_output_parallel()
  File "/home/macvo/workspace/Module/Optimization/Interface.py", line 147, in __get_output_parallel
    assert self.child_proc and self.child_proc.is_alive()
AssertionError
macvo@235c0f0056a9:~/workspace$ 
```

closed 2025-07-15 by MarkChenYutian

LVIC11, 2025-10-17:
@ddkats thanks for reply!i have solved it through updating  nvidia-drivers,maybe it works.:)

ddkats, 2025-10-17:
@LVIC11 Do you mean the drivers themselves or just the NVIDIA toolkit? In other words, did you change the version shown by nvidia-smi (drivers) or by nvcc -V (toolkit)? 

Also, which drivers are you using? I’m on 12.2 for both the drivers-toolkit.

LVIC11, 2025-10-17:
@ddkats If you have any questions,plz send me to 2208348810@qq.com :)