MAC-VO/MAC-VO #30: the relationship between pixels, depth, and three-dimensional coordinates

aidadgy, 2025-08-16:
Hello author，in your paper, you mentioned the relationship between pixels, depth, and three-dimensional coordinates. However, strictly following the coordinate conversion formula, **di,t** in the following formula should be **zi, t**. Why do you make **zi, t** equal to **di,t**
file:///home/alading/%E5%9B%BE%E7%89%87/2025-08-16%2017-46-50%20%E7%9A%84%E5%B1%8F%E5%B9%95%E6%88%AA%E5%9B%BE.png

MarkChenYutian, 2025-08-21:
Hi @aidadgy yes we are using the NED convention for the code implementation, but when deriving formula we didn't strictly follow that coordinate convention... If strictly following the code, it should be x_i, t = d_i, t since we are using NED convention.

MarkChenYutian, 2025-08-22:
Hi @aidadgy I'm closing this issue due to inactivity, feel free to reopen if you have more followup questions.

closed 2025-08-22 by MarkChenYutian